techandme/NextBerry #11: Install errors

ezraholm50, 2017-01-19:
@enoch85 

```
Package open-vm-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  open-vm-tools-dkms

E: Package 'open-vm-tools' has no installation candidate
gpg: directory `/root/.gnupg' created
gpg: new configuration file `/root/.gnupg/gpg.conf' created
gpg: WARNING: options in `/root/.gnupg/gpg.conf' are not yet active during this run
gpg: keyring `/root/.gnupg/secring.gpg' created
gpg: keyring `/root/.gnupg/pubring.gpg' created
gpg: requesting key A724937A from hkp server p80.pool.sks-keyservers.net
gpg: /root/.gnupg/trustdb.gpg: trustdb created
gpg: key A724937A: public key "Nextcloud Security <security@nextcloud.com>" imported
gpg: no ultimately trusted keys found
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
gpg: can't open `/tmp/gpg/nextcloud-.zip.asc'
gpg: verify signatures failed: file open error
Package NOT OK! Installation is aborted...

```

ezraholm50, 2017-01-19:
`curl -s https://download.nextcloud.com/server/releases | tac | grep unknown.gif | sed 's/.*"nextcloud-\([^"]*\).zip.sha512".*/\1/;q'`

Output null

ezraholm50, 2017-01-19:
```
root@Nginx:~# curl -s https://download.nextcloud.com/server/releases
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>301 Moved Permanently</title>
</head><body>
<h1>Moved Permanently</h1>
<p>The document has moved <a href="https://download.nextcloud.com/server/releases/">here</a>.</p>
</body></html>

```

enoch85, 2017-01-19:
```
"curl -s https://download.nextcloud.com/server/releases"
``` 

Works here...

```
curl -s https://download.nextcloud.com/server/releases | tac | grep unknown.gif | sed 's/.*"nextcloud-\([^"]*\).zip.sha512".*/\1/;q'
```

Yes?

```
E: Package 'open-vm-tools' has no installation candidate
gpg: directory `/root/.gnupg' created
gpg: new configuration file `/root/.gnupg/gpg.conf' created
gpg: WARNING: options in `/root/.gnupg/gpg.conf' are not yet active during this run
gpg: keyring `/root/.gnupg/secring.gpg' created
gpg: keyring `/root/.gnupg/pubring.gpg' created
gpg: requesting key A724937A from hkp server p80.pool.sks-keyservers.net
gpg: /root/.gnupg/trustdb.gpg: trustdb created
gpg: key A724937A: public key "Nextcloud Security <security@nextcloud.com>" imported
gpg: no ultimately trusted keys found
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
gpg: can't open `/tmp/gpg/nextcloud-.zip.asc'
gpg: verify signatures failed: file open error
Package NOT OK! Installation is aborted..
``` 

Works here...

ezraholm50, 2017-01-19:
I wonder if it has to do with any dependency's of ARM.
Other than that its coming along quite quickly! Should be done in about a week.

enoch85, 2017-01-19:
Try it out on a N-VM. You'll see it works. :)

ezraholm50, 2017-01-19:
I've tried it on several Ubuntu servers, removing -s gives no output. ARM (RPI) fails here in the script.

enoch85, 2017-01-19:
It's a variable... set -x and see the outcome.

ezraholm50, 2017-01-19:
```
root@Nginx:~# curl -x https://download.nextcloud.com/server/releases | tac | grep unknown.gif | sed 's/.*"nextcloud-\([^"]*\).zip.sha512".*/\1/;q'
curl: no URL specified!
curl: try 'curl --help' or 'curl --manual' for more information

```

ezraholm50, 2017-01-19:
```NCVERSION=$(curl -s --max-time 900 $NCREPO/ | tac | grep unknown.gif | sed 's/.*"nextcloud-\([^"]*\).zip.sha512".*/\1/;q')```

Fix!

closed 2017-01-19 by ezraholm50