techandme/NextBerry #18: Key check

ezraholm50, 2017-01-21:
```
gpg: directory `/root/.gnupg' created
gpg: new configuration file `/root/.gnupg/gpg.conf' created
gpg: WARNING: options in `/root/.gnupg/gpg.conf' are not yet active during this run
gpg: keyring `/root/.gnupg/secring.gpg' created
gpg: keyring `/root/.gnupg/pubring.gpg' created
gpg: requesting key A724937A from hkp server p80.pool.sks-keyservers.net
gpg: /root/.gnupg/trustdb.gpg: trustdb created
gpg: key A724937A: public key "Nextcloud Security <security@nextcloud.com>" imported
gpg: no ultimately trusted keys found
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
gpg: Signature made Sun 15 Jan 2017 04:07:44 PM UTC using RSA key ID A724937A
gpg: Good signature from "Nextcloud Security <security@nextcloud.com>"
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 2880 6A87 8AE4 23A2 8372  792E D758 99B9 A724 937A
Package OK!
```

@enoch85

enoch85, 2017-01-22:
Yes? It's working. :)

ezraholm50, 2017-01-22:
Yeah it works but these messages shouldn't be seen in such a check, I assume, right?
```
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
```

enoch85, 2017-01-23:
cc @lukasreschke

ezraholm50, 2017-02-02:
@lukasreschke Any update please, issue still persist. (not really an issue but I don't think it should be there)

enoch85, 2017-05-17:
This is fixed now in lib.sh there's a function.

closed 2017-05-17 by ezraholm50