techandme/NextBerry #21: dead download link for NextBerry

ssklare, 2017-01-29:
Hello, 
As of now the download link for NextBerry appears to be dead.
https://cloud.techandme.se/s/uMwUovc0iPn9DcA
Looking forward to trying out NextBerry,
Thank you

## Steps to reproduce
1.
2.
3.

## Expected behaviour
Tell us what should happen

## Actual behaviour
Tell us what happens instead

## Server configuration

**Server version:** (see your admin page)

**Raspberry 2 or 3?**

**Updated from an older installed version or fresh install:**

## Logs

#### Log file (/var/ncdata/nextcloud.log)
```
Insert your nextcloud.log file here
```

#### Installation log (command output)
```
Insert the CLI output here
```

ezraholm50, 2017-01-30:
@enoch85 My server is up and all the links check out could you investigate.

Here's a mirror: https://cloud.waaromzomoeilijk.nl/s/dxFxgOqs6crjQYe

enoch85, 2017-01-30:
It's gone from my server. The whole folder. Did you remove the share?

Anyway, share it from your server instead. I don't have time to fix it now.

There is something wrong with your federation it seems, as I can share 
links but mine doesn't show up when I share tih you.

Best Regards,
Daniel Hansson
*Tech and Me <https://www.techandme.se/>*

enoch85, 2017-01-30:
I just uploaded a new folder, share to that please.

Best Regards,
Daniel Hansson
*Tech and Me <https://www.techandme.se/>*

ezraholm50, 2017-01-30:
Thanks though I can't figure out the new shared links. Will do it later today.

enoch85, 2017-01-30:
Its in your Root in your cloud.

enoch85, 2017-01-30:
@ezraholm50 I will be on IRC later tonight. We really have to sort this, it's very strange that you can't see my shared federation links.

ezraholm50, 2017-01-30:
Not sure if I have time tonight. Links are all temporary switched to my
server.

ezraholm50, 2017-01-30:
My 2 cents is that you clone the repo of my federated share so we have 2
independent mirrors instead of 2 mirrors of one linked to the other. (Can't
edit on mobile)

enoch85, 2017-01-30:
@ezraholm50 I'm online now.

closed 2017-01-30 by enoch85

ezraholm50, 2017-01-30:
Switched to my server as primary download link. Github + techandme.se.
All tested and working.

ezraholm50, 2017-01-30:
@ssklare You're welcome, please if you find the time post your findings/logs in the issue section so we can improve even more!