techandme/NextBerry #41: Installation hangs?

paarsand, 2017-02-10:
## Steps to reproduce
1. Install to sd
2.
3.

## Expected behaviour
installation should finish

## Actual behaviour
installation hangs, see screenshot 
[image]
i have tried installing only to SD, and SD / HARDDRIVE

If i try to type something at this point i get invalid email-adress.
If i reboot, it get to the nextberry statuspage with cpu temps and ip adresses, but no commands or webserver works. 

## Server configuration

**Raspberry 2 or 3?** 3 
3
**Updated from an older installed version or fresh install:** fresh

paarsand, 2017-02-10:
Ah, i think there is something wrong in the script, what really should appear here is
[image]

i got it running by running the next script manually, but my installation is borked, something is missing.

enoch85, 2017-02-10:
@ezraholm50 FYI

@paarsand and me discussed this on FB. I don't think it's an error with the scripts, just that the RPI is toooo slow and there is a lot of latency. 

Did you run the Let's Encrypt scripts successfully @ezraholm50?

paarsand, 2017-02-10:
Ok, thought i should leave you guys a note. when my server hangs at the point of creating log. i can input my email, and press "a" and "n" like the second screenshot, and the installation continues....

There is a bug displaying the text. entering it without guideline and the setup continues. 

I got a tip of waiting, i waited 4 hours, so the wait is not worth it. tried a new install, and immideately typed in my email and a, n. and presto! it works. :)

paarsand, 2017-02-10:
Well, installation finished, rebooted. now my rpi3 is not booting. 

Guess the happiness was short lived, but everything continued as it should and setup completed. With every setting, feature enabled. But now i cannot contact my rpi. 

Guess i'll be sufficent with my Virtualbox-VM until this one works :P

ezraholm50, 2017-02-11:
Sorry for the delay. @enoch85 haven't tried the letsencrypt script. Will give it a run here at home.

@paarsand keep shift pressed during boot and see what it does. Note that I haven't ran it on a PI3 yet but it should just work as only config.txt in /boot is different. 
In order for the onboard wifi/bluetooth to work we might need to adjust some stuff. 

When its online please upload the install log (the login screen will inform you how to)

Else if that doesn't work please re run the install but first remove: reboot from the nextcloud-startup-script.sh and then upload the logs so we can investigate.

Offtopic: did you manage to overclock the PI3 already?

paarsand, 2017-02-12:
I have not tried overclocking, and i don't really think i want to bother. having enough trouble as is.

I cannot get this bugger to work, i've got everything to finish, with entering email info when the script freaks out. But since this is headless i don't see whats going on. (i do currently not have a screen, only my laptop).

ezraholm50, 2017-02-12:
Aah please get it going on a screen with a keyboard so we can debug.

enoch85, 2017-02-12:
@paarsand Edit the script and enter Debug mode. It's a switch that you turn on in the top.

Or you can write 'set -x' in the top of eah script.

ezraholm50, 2017-02-12:
First try if holding shift during boot works. Then try to upload the logs like I said earlier. If that all doesn't work then try to reinstall the entire NextBerry with debug mode on like enoch85 said.

enoch85, 2017-02-12:
...and @paarsand Thank you very much for taking your time doing this!

paarsand, 2017-02-15:
Ok, screen attached!

I did install YET another time, getting used to this now. crawled through the letsencrypt bug. 
reboot, and dead. So i coupled this bad boy on to my Living room TV.

[image]
The box does not even try to get an ip, so when i hooked it up to my TV, this is displayed (i do not have network over at the tv's place) 

Then i logged in, and a new login appeared. this one was lagging badly. 
[image]

Nothing happened here, so i ctrl+c'ed, tried to ran nextcloud.sh, witch freaked out totally, and now i'm lost again, because this bugger wont even try to get an internetconnection. btw, sudo wireless is dead.. :(

[image]

I'm trying to exctract logs, but where the hell do i find them?

paarsand, 2017-02-15:
Installation absolutely wrecked the installation, nothing works. the sudo commands on screen does not work, eth0 has crashed, does not get ip anymore. I wish i could get this bugger online, and i would gladly give you my root password for error-checking.

enoch85, 2017-02-15:
@ezraholm50 I suggest you update the images since we switched repo.

paarsand, 2017-02-15:
Yeah, that another ting i forgot. the images are wrecked after the repo-update. i had to both edit and manually download the images.

But the very un-cool thing is that network is borked after the last reboot, when letsencrypt crashes..

enoch85, 2017-02-15:
@paarsand What's your which subnet do you use? This seems to get a 10.X.X.X address

paarsand, 2017-02-15:
That is correct, 10.0.0.95 is the static ip to this. not just static, but incase dhcp i've locked the mac to 95 in dhcp. so ip is correct.

I logged in to the router, the client does not appear, and dissapears after the letsencrypt-script, then the client dies.

Is letsencrypt the last script? Or is anything supposed to run after that reboot?

enoch85, 2017-02-15:
Let's Encrypt is the last one that is run. What happens if you skip that, do you get connectivity?

paarsand, 2017-02-15:
i dont know, but i'm about to try.

enoch85, 2017-02-15:
Wow, you're just great! Our new bug helper. :1st_place_medal:

paarsand, 2017-02-15:
i cannot use this connected to my tv, as i just reallised i do not have cabled internet over there. so i figured i could do the sudo wireless

paarsand, 2017-02-15:
I can be bug helper, but please fix those images, because its hell on earth to get the right scripts after you changed it.

enoch85, 2017-02-15:
We are on it. Ping @ezraholm50

paarsand, 2017-02-15:
can i get a logfile from letsencrypt? and in case where is it located?

enoch85, 2017-02-15:
You can always install both Let's Encrypt and Collabora afterwards. ;)

paarsand, 2017-02-15:
Fresh install, without ANY options, not lets encrypt or anything. "install suceeded" it says, reboots, and no eth0.

Everything works ok until the final reboot, then network dies. Again. :(

enoch85, 2017-02-15:
@ezraholm50 You really need to take a look at this.

ezraholm50, 2017-02-16:
Sorry for the absence. Please run through the scripts and before the last script starts (nextcloud-startup) CTRL+C edit the file and remove the reboot at the bottom of the file.

When the installation now ends try to upload the logs and/or paste the content of /etc/network/interfaces please.

paarsand, 2017-02-17:
I'm sorry, but i do not really have the time to try to correct this. I would be very happy if you, that have the knowledge would get an pi3 to try to correct this, but i do not have the time or patience for it :(

ezraholm50, 2017-02-18:
I'd be happy to if anyone would provide the hardware. I'd cover the
shipping.

enoch85, 2017-02-22:
@ezraholm50 Please rebuild the images and release v1.3 thank you!

ezraholm50, 2017-02-25:
No need to rebuild them. 1.3 is already in master. Have no time right now
few days I'll be back home.

paarsand, 2017-02-25:
Do you think this will Solve my issue?

ezraholm50, 2017-02-25:
No it will not, these are just minor changes to the script and/or
improvements from the VM branch. Not much time right now to look at it.
Will report back when a solution is found.

ezraholm50, 2017-05-17:
Alright Repo updated. Could use some help testing. @paarsand

paarsand, 2017-05-17:
Any link to updated image?

ezraholm50, 2017-05-17:
Nothing wrong with the image, its a base Ubuntu 16.04 server with only a user added and a login script.
https://github.com/techandme/NextBerry/releases/download/V1.2/NextBerry.img.zip

Note that something could go wrong, haven't tested it yet. Just implemented all changes (105) commits from master nextcloud/vm.

This weekend I'll do a test run myself, you could wait until then or help test it with us.

paarsand, 2017-05-18:
I have a busy weekend, but if i get the time i'll test it :)

Will follow this one. currently i'm running my own setup, but its not working nearly as good, so i'm watching this one :)

ezraholm50, 2017-05-27:
Getting close to a stable release...

ezraholm50, 2017-06-10:
Stable, closed

closed 2017-06-10 by ezraholm50