techandme/NextBerry #47: Can't update NextBerry

OskarNadolny, 2017-03-28:
## Steps to reproduce
1. Download img from here https://cloud.techandme.se/s/G6PaI0miBibhDwj
2. Install NextBerry
3. Try update by executing "sudo bash /home/ncadmin/nextberry-upgrade.sh"

## Expected behaviour
Should update

## Actual behaviour
```
ncadmin@ubuntu-standard:~$ sudo bash /home/ncadmin/nextberry-upgrade.sh
11 already applied...
nextberry-upgrade.sh: line 115: syntax error near unexpected token `else'
nextberry-upgrade.sh: line 115: `else'
```

## Server configuration

**Server version:** i don't know how to check it

**Raspberry 2 or 3?**: Rpi 2 

**Updated from an older installed version or fresh install:** fresh install

## Logs

#### Log file (/var/ncdata/nextcloud.log)
```
Insert your nextcloud.log file here
```

#### Installation log (command output)
```
cadmin@ubuntu-standard:~$ sudo install-log
Failed to contact the server: [Errno socket error] The write operation timed out
```

enoch85, 2017-05-07:
Thanks for your report and sorry for the late response! This is @ezraholm50's repo and he is away at the moment.

ezraholm50, 2017-05-07:
Hardware broke down can't test it anymore. No PI's around to do testing. Also as of the commits from upstream VM the install script is broken. Those commits haven't been tested on the ARM build.

enoch85, 2017-05-07:
@ezraholm50 so what if I bought you a Pi3 would you be able to test then?

ezraholm50, 2017-05-07:
That wouldn't be a problem. That way we can also fully support PI 3. Only
need the power supply if its other than 2 amps 5v. Got all the other stuff
around.

enoch85, 2017-05-07:
@ezraholm50 Ok so tell me what you need and I'll arrange it for you. Please send me an email with your shipping address and the details. Thanks!

enoch85, 2017-05-07:
@ezraholm50 I need your address, I found a place in Holland that I buy the Pi3 and send it to you directly.

ezraholm50, 2017-05-07:
Sos solutions? If so i can pick it up there...

enoch85, 2017-05-07:
@ezraholm50 I still need your address, Huisnummer for example? Send me an email.

enoch85, 2017-05-07:
Ok so it's ordered. Finally we can get this repo going again! :)

ezraholm50, 2017-05-17:
Ok repo is now fully up to date with current master of the VM. Could you try to re-run the script.

ezraholm50, 2017-05-17:
I'll close this as I've seen the error message you posted in the ShellCheck and it has been resolved.
https://travis-ci.org/techandme/NextBerry

Please reopen if the issue persists. (Please re-download the upgrade script and run it)

closed 2017-05-17 by ezraholm50