techandme/NextBerry #48: Can't install NextBerry

voodoo42, 2017-05-14:
## Steps to reproduce
1. Downloaded and copied to SD-Card like described (File: https://github.com/techandme/NextBerry/releases/download/V1.2/NextBerry.img)
2. Raspberry ist booting, login with ncadmin/nextcloud
3. another login fpr "sudo" with nextcloud

## Expected behaviour
Ongoing instalation

## Actual behaviour
Errorscreen like photo in this post "...Temporary failure in name resolution..."

## Server configuration

**Server version:** (see your admin page)

**Raspberry 2 or 3?** RPi 3

**Updated from an older installed version or fresh install:**

## Logs

#### Log file (/var/ncdata/nextcloud.log)
```
Insert your nextcloud.log file here
```

#### Installation log (command output)
```
Insert the CLI output here
```
[image]

ezraholm50, 2017-05-14:
Thanks for your post. First, Are you connected via Ethernet cable? 

Could you post the out put of: `sudo ifconfig; sudo cat /etc/network/interfaces; sudo ifconfig -a`

voodoo42, 2017-05-16:
Many thanks for the quick answer. 
I got, noob problem. I'm very sorry.
I think we can close this.

closed 2017-05-16 by voodoo42

ezraholm50, 2017-05-16:
No problem, glad that its resolved.