techandme/NextBerry #53: Startupscript utf mysql fails

ezraholm50, 2017-05-21:
@enoch85 Any idea?

```
Enabling UTF8mb4 support on nextcloud_db....
Please be patient, it may take a while.
[/] .....] Restarting mysql (via systemctl): mysql.service
System config value mysql.utf8mb4 set to boolean true
 - Repair MySQL collation
     - Change row format for oc_accounts ...
     - Change collation for oc_accounts ...
     - Change row format for oc_activity ...
     - Change collation for oc_activity ...
     - Change row format for oc_activity_mq ...
     - Change collation for oc_activity_mq ...
     - Change row format for oc_addressbookchanges ...
     - Change collation for oc_addressbookchanges ...
     - Change row format for oc_addressbooks ...
     - Change collation for oc_addressbooks ...

                                                                                                                                
  [Doctrine\DBAL\Exception\DriverException]                                                                                     
  An exception occurred while executing 'ALTER TABLE `oc_addressbooks` CONVERT TO CHARACTER SET utf8mb4 COLLATE utf8mb4_bin;':  
  SQLSTATE[42000]: Syntax error or access violation: 1071 Specified key was too long; max key length is 767 bytes               
                                                                                                                                

                                                                                                                   
  [Doctrine\DBAL\Driver\PDOException]                                                                              
  SQLSTATE[42000]: Syntax error or access violation: 1071 Specified key was too long; max key length is 767 bytes  
                                                                                                                   

                                                                                                                   
  [PDOException]                                                                                                   
  SQLSTATE[42000]: Syntax error or access violation: 1071 Specified key was too long; max key length is 767 bytes  
                                                                                                                   

maintenance:repair [--include-expensive]

Sorry but something went wrong. Please report this issue to https://github.com/nextcloud/vm/issues and include the output of the error message. Thank you!
sudo -u www-data /var/www/nextcloud/occ maintenance:repair failed

```

enoch85, 2017-05-21:
Which MySQL version? Maybe it differs between 5.7 and 5.X?

Also check the Nextcloud documentation: https://docs.nextcloud.com/server/11/admin_manual/maintenance/mysql_4byte_support.html

ezraholm50, 2017-05-21:
After: `sudo -u www-data /var/www/nextcloud/occ config:system:set mysql.utf8mb4 --type boolean --value="false"`

and maintenance:repair it functions again. Will figure out how to implement utf8mb4 support. Thanks.