techandme/NextBerry #73: Installation Fails

TheCrey, 2017-05-27:
## Steps to reproduce
1. Start Installation -> after the first reboot 
2.
3.

## Expected behaviour
Installation should run through

## Actual behaviour
It stops with the error message below

## Server configuration

**Server version:** not able to say as installation does not Finnish

**Raspberry 2 or 3?** 3

**NextBerry version** not able to say as installation does not Finnish

**Updated from an older installed version or fresh install:** fresh install 

## Network
**Do you use DHCP?** yes but raspberry gets always the same ip 

**Is port 80 and/or 443 open?** yes

## Logs / Screenshots

Here are the Logs i get 

https://hastebin.com/irezifubeh.sql

TheCrey, 2017-05-27:
Hope this helps, i tried i to now 3 times, same procedure, same issue. 

Any ideas ?

ezraholm50, 2017-05-27:
Yes thanks working hard on fixing this.  Has to do with repo's. Php 7.0 has
a 3rd party repo on rasbian. Working on fixing this, release 1.4 will fix
this. I will keep you updated.

TheCrey, 2017-05-27:
Thanks, thats nice ti hear, if i can do something let me know, i am willing to help. 

Regards
Crey

ezraholm50, 2017-05-27:
Thanks, main problem is the package priority... i want all packages except
php to be grabbed out of jessie stable and php from jessie-backports 3rd
party repo. Now it installs but in the end it doesn't get installed, even
though is fully installs as far as i can see.

ezraholm50, 2017-05-27:
Alright, had something to do with php-smbclient, while running apt-get install -y php-smbclient had some errors and with -y specified it removed all php7.0 packages. Took php-smbclient out for now, will implement it on a nextberry-upgrade soon! 

Please redownload the image, its cleaner only added a sudo user. (view readme for link)

closed 2017-05-27 by ezraholm50

TheCrey, 2017-05-27:
Hi, 

thanks for that fast action. 

I think i may be blind, but i am missing the download link, its gone.. i only see the source code zip, not the img file anymore

ezraholm50, 2017-05-27:
@TheCrey you're welcome! Sorry, forgot to submit... https://github.com/techandme/NextBerry/releases/tag/Images