techandme/NextBerry #76: run static script

ezraholm50, 2017-05-27:
@enoch85 are you redirecting the output to a file or what is the function of >> $SCRIPTS ?

```
# Dirs
SCRIPTS=/var/scripts

# Run any script in ../static
# call like: run_static_script name_of_script
run_static_script() {
    # Get ${1} script
    rm -f "${SCRIPTS}/${1}.sh" "${SCRIPTS}/${1}.php" "${SCRIPTS}/${1}.py"
    if wget -q "${STATIC}/${1}.sh" -P "$SCRIPTS"
    then
        bash "${SCRIPTS}/${1}.sh >> $SCRIPTS"
        rm -f "${SCRIPTS}/${1}.sh"
    elif wget -q "${STATIC}/${1}.php" -P "$SCRIPTS"
    then
        php "${SCRIPTS}/${1}.php"
        rm -f "${SCRIPTS}/${1}.php"
    elif wget -q "${STATIC}/${1}.py" -P "$SCRIPTS"
    then
        python "${SCRIPTS}/${1}.py"
        rm -f "${SCRIPTS}/${1}.py"
    else
        echo "Downloading ${1} failed"
        echo "Script failed to download. Please run: 'sudo wget ${STATIC}/${1}.sh|php|py' again."
        sleep 3
    fi
}
```

Because the installation fails to run the script:
```
+ run_static_script change-ncadmin-profile
+ rm -f /var/scripts/change-ncadmin-profile.sh /var/scripts/change-ncadmin-profile.php /var/scripts/change-ncadmin-profile.py
+ wget -q https://raw.githubusercontent.com/techandme/NextBerry/master/static/change-ncadmin-profile.sh -P /var/scripts
+ bash '/var/scripts/change-ncadmin-profile.sh >> /var/scripts'
bash: /var/scripts/change-ncadmin-profile.sh >> /var/scripts: No such file or directory
+ rm -f /var/scripts/change-ncadmin-profile.sh
```

enoch85, 2017-05-27:
Strange! You can the what the codes does...

But why `>> /var/scripts`? something you added?

ezraholm50, 2017-05-27:
No it was there by default, just a pull req from upstream vm? I've removed it and it works...

closed 2017-05-27 by ezraholm50