techandme/NextBerry #82: Installation -> 1st setup fails

TheCrey, 2017-05-27:
## Steps to reproduce
1. Install from SD -> first fresh installation 
2. still default ncadmin user, never changes any password 
3. no mysql password set 

## Expected behaviour
Installation works

## Actual behaviour
installation stops 

## Server configuration

**Server version:** 
Version 12 

**Raspberry 2 or 3?**
Raspberry 3

**NextBerry version**
n/a

**Updated from an older installed version or fresh install:**
fresh install 

## Network
**Do you use DHCP?**
Yes, raspberry gets always same ip 

**Is port 80 and/or 443 open?**
booth open 

## Logs / Screenshots
This is a new screen in the console, so i get only this error. 
If i restart the script, as described i run into the same error 

https://hastebin.com/aweriwiwec.vbs

TheCrey, 2017-05-27:
running the update sh does not solve this issue 

`root@nextberry:~# sudo bash /var/scripts/update.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Checking latest released version on the Nextcloud download server and if it's possible to download...
[|] Please wait... SUCCESS!
Latest version is: 12.0.0. Current version is: 12.0.0.
No need to upgrade, this script will exit...
root@nextberry:~# `

ezraholm50, 2017-05-28:
@TheCrey fixed these MySQL errors in https://github.com/techandme/NextBerry/pull/83 and https://github.com/techandme/NextBerry/pull/84

closed 2017-05-28 by ezraholm50