techandme/NextBerry #97: Pi boots to emergency, can't login

newkindofworld, 2017-05-30:
Hi !
Complete noob here, tried to get it to work on my Pi 3.
The pi boots to emergency mode, can't login ncadmin as account doesn't seem to exist ... don't know what to do ! 
See picture in bottom

## Steps to reproduce
1. Fresh install
2. Boot
3. Get to emergency mode

## Expected behaviour
Should be able to login and complete initial setup ?

## Actual behaviour
Stuck on emergency mode

**Raspberry  3**

**fresh install:**

## Network
**Do you use DHCP? yes**

**Is port 80 and/or 443 open? yes**

#### Screenshoots
http://imgur.com/a/IOu77

ezraholm50, 2017-05-30:
Thanks. Try recreating the sd card again, do you use windows?

If the same thing happens try holding shift

newkindofworld, 2017-05-30:
Did it 3 times, using Win32 Disk Imager in admin mode, on two different SD cards (samsung evo + 32gb and 64gb).

ezraholm50, 2017-05-30:
Mount the sd on your pc try to change /boot/config.txt

find arm_freq=1450 
change to arm_freq=1200

then boot again

newkindofworld, 2017-05-30:
Just checked, seems already to be set @ 1200, got this in config.txt :

##
[pi2]
arm_freq=1000
[pi3]
arm_freq=1200
[all]

##

ezraholm50, 2017-05-30:
Weird could you perhaps try it on a linux box or with a different program.
It works fine here.. just tried it.

Do you get the resize prompt? What download link did you use? Try
downloading it again.

newkindofworld, 2017-05-30:
Ok at least it's reassuring !

No resize prompt ever, I used the second donwload link in here https://www.techandme.se/nextberry-rpi/ , the first one doesn't work.

I'll try to get my hands on a linux machine, even if I'm not quite familiar with those :)

Thanks for your help, I'll keep you updated !

ezraholm50, 2017-05-30:
Go no further please use this image. The one you have is old and broken. Tha ks for reporting.

https://github.com/techandme/NextBerry/releases/tag/Images

ezraholm50, 2017-05-30:
Fixes link is changed. Let me know if it worked out for you please.

closed 2017-05-30 by ezraholm50

newkindofworld, 2017-05-30:
Worked with the good link / good image, thanks for your help :)

ezraholm50, 2017-05-30:
No problem glad to help.