techandme/NextBerry #99: Setup crashing

newkindofworld, 2017-06-08:
## Steps to reproduce
1. Created SD using Win32 Disk Imager (Samsung 64 Gb evo+)
2. First boot, then expanded system to full size of sd
3. Crashes after informing that creating a new user would be optimal

## Expected behaviour
setup should continue ?

## Actual behaviour
Error (see picture)

## Server configuration

**Server version:** (see your admin page)

**Raspberry 3**

**NextBerry latest**

**fresh install:**

## Network
**Do you use DHCP? yes**

**Is port 80 and/or 443 open? yes**

#### Screenshoots

ezraholm50, 2017-06-08:
Could you link me to where you downloaded the image? Have you tried recreating the image on the SD? If that doesnt work try another SD card.

If that also doesnt solve your trouble, try holding shift every time it boots until the login screen. Please keep me updated.

newkindofworld, 2017-06-09:
Sorry, I was a bit busy.
Downloaded from here : https://github.com/techandme/NextBerry/releases/tag/Images
I tried 3 times recreating the image, on same SD, crashes at different times. Tried another SD, same result.
Holding shift helped me to create the new sudo user, but then crashed with visual glitch on screen.
I'm beginning to suspect my Pi 3 is faulty, will try another one later.
I'll keep you updated.
Thanks for your help !

ezraholm50, 2017-06-09:
Ive never seen this error before. Broadcom kernel stuff... I test everything on my PI3 l, Ill run it again later today to see if i can reproduce. 

Good one.. it could be your PI, do you have another one around?

Holding shift only delays the error?

newkindofworld, 2017-06-09:
See the kind of errors I'm having now : 
http://imgur.com/a/1HDmg
http://imgur.com/a/6a5jd
I have to other Pi at home, I'm conducting tests at work, so I'll try in the next few hours.
Will get other SDs and power supply too, never know what could be faulty !

Holding shift only delays the errors indeed.

ezraholm50, 2017-06-09:
Wow thats a rainbow of new errors. Id have to google around this weekend to check it out.

Mostly kernel panics are due to the overclocking i could try to create a custom image for you...

Please keep me updated ill do the same

newkindofworld, 2017-06-09:
Never overclocked, don't even know how to do it.
I'll keep you updated with my new bunch of tests later today.
Thanks again !

newkindofworld, 2017-06-09:
Nevermind, the Pi is somehow faulty, full setup runs flawlessly on another one.
Weird, as the first one is running fine a recalbox distro, retrogaming emulation and all; raspbian jessie too.
Thanks for your help, and sorry for the waste of time.

ezraholm50, 2017-06-09:
Thats weird indeed, no problem thank you for reporting.

Not every PI is the same this is more so with overclocking some can handle it and some cant.

When the installation finishes You will be presented with a nice welcome screen and you can type: sudo install-log

This will give you a link to the installation log url please post it. That enables me to filter our bugs etc.

Glad to help, if any issues arise please let me know.

ezraholm50, 2017-06-09:
Thank you very much. Does it auto redirect to https and does the nextcloud background image show in the browser?

newkindofworld, 2017-06-10:
For the moment, it just advises me to enable https in administration settings, and I don't have yet the knowledge to do so (don't know where is the vhost config file to edit), I will go on with setup in the next few days.

ezraholm50, 2017-06-10:
Thanks so the auto redirect is not working. Just add https:// before your url/ip in the browser and its all good.

newkindofworld, 2017-06-10:
Simple indeed :)
I manage to add my dynamic dns domain to the trusted domains, so I can remote access nextcloud, it seems I have no background image at login form.
http://imgur.com/a/WESaR

ezraholm50, 2017-06-10:
Yes thats a byproduct of the https error ill create a fix soon

ezraholm50, 2017-06-10:
Please run:
`sed -i 's|  RewriteRule ^(.*)$ https://%{HTTP_HOST}%{REQUEST_URI} [L,R=301]||g' /var/www/nextcloud/.htaccess`

and:
`sed -i 's|  RewriteCond %{HTTPS} off||g' /var/www/nextcloud/.htaccess`

Then check in the browser if via https://url or ip the background image shows.

enoch85, 2017-06-10:
Fir some strange reason you need a valid SSL cert for the background image to show. Same error on the VM. Run LE to sort it out.

ezraholm50, 2017-06-10:
Ouch thats weird, I'd like to auto rewrite to https but adding this in .htaccess doesnt work for some reason. I think i'll add it to vhost:80

Last time i checked it was because of the images are beeing served over http instead of https...

Also just using snakeoil certs the background does display on LAN, on the PI.

newkindofworld, 2017-06-10:
I did not setup LE as it was a first try, and I don't have a domain; is it working with ddns like no-ip ?
And if so, how can I run LE again ?

ezraholm50, 2017-06-10:
Thats alright, you can use your ddns domain. You won't HAVE to use lets encrypt you can just add an exception for the current SSL cert.

If you want to try out lets encrypt run this:
sudo rm /var/scripts/activate-ssl.sh

sudo wget https://github.com/techandme/NextBerry/raw/beta/lets-encrypt/activate-ssl.sh -P /var/scripts/

sudo bash /var/scripts/activate-ssl.sh

@enoch85 this should be enough I think?
```
    cat << HTTP_CREATE > "$HTTP_CONF"
<VirtualHost *:80>

### YOUR SERVER ADDRESS ###
#    ServerAdmin admin@example.com
#    ServerName example.com
#    ServerAlias subdomain.example.com

### SETTINGS ###
    DocumentRoot $NCPATH

    <Directory $NCPATH>
    Options Indexes FollowSymLinks
    AllowOverride All
    Require all granted
    Satisfy Any
    </Directory>

    <IfModule mod_dav.c>
    Dav off
    </IfModule>

    <Directory "$NCDATA">
    # just in case if .htaccess gets disabled
    Require all denied
    </Directory>

    SetEnv HOME $NCPATH
    SetEnv HTTP_HOME $NCPATH

    RewriteEngine On
    RewriteCond %{HTTPS} off
    RewriteRule ^(.*)$ https://%{HTTP_HOST}%{REQUEST_URI} [L,R=301]

</VirtualHost>
HTTP_CREATE
```

newkindofworld, 2017-06-10:
Just did the LE script, everything went fine, but now I can't access nextcloud web interface, locally or remotely, http error 500 using chrome.
(LE cert seems ok, as firefox shows the connection is secured using a LE cert)

ezraholm50, 2017-06-10:
What domain did you use? Please post the output of: `cat /var/log/apache2/error.log`
Also reboot and see if that makes a difference. Also use https://yourdomain.com and try https://yourIPwan

newkindofworld, 2017-06-10:
my dynamic dns is a ddns.net 
here's the output : https://pastebin.com/X73sdryP
Already rebooted twice.
Tried urls starting in https , and even on my wan ip directly, doesn't work, error.

ezraholm50, 2017-06-10:
Please post the output of: `cat /etc/apache2/sites-available/nextcloud_ssl_domain_self_signed.conf`

To see if it actually changed your domains, also post the out put of: `cat /var/log/syslog`
But use pastebin for this please.

newkindofworld, 2017-06-10:
https://pastebin.com/86SzXG60
syslog is big, pasted it in 3 parts : 
https://pastebin.com/AtAmj7Mw
https://pastebin.com/4YiJRS9c
https://pastebin.com/MF9gGLRy

ezraholm50, 2017-06-10:
Ok apache2 doesn't have your new certs yet. I'll post some how to's tonight ok?

newkindofworld, 2017-06-10:
sure, I'm not in a hurry, whenever you have time, thanks again !

ezraholm50, 2017-06-10:
Please post the output of: `ls /etc/letsencrypt/live; ls /etc/certbot/live`

Glad to help!

newkindofworld, 2017-06-10:
root@synojerem:/home/ncadmin# ls /etc/letsencrypt/live; ls /etc/certbot/live
synojerem.ddns.net
ls: cannot access /etc/certbot/live: No such file or directory

ezraholm50, 2017-06-10:
Please run this:

`sed -i 's|/etc/ssl/certs/ssl-cert-snakeoil.pem|/etc/letsencrypt/live/synojerem.ddns.net/cert.pem|g' /etc/apache2/sites-available/nextcloud_ssl_domain_self_signed.conf`

`sed -i 's|/etc/ssl/private/ssl-cert-snakeoil.key|/etc/letsencrypt/live/synojerem.ddns.net/privkey.pem|g' /etc/apache2/sites-available/nextcloud_ssl_domain_self_signed.conf`

`service apache2 restart`

Then try to visit:
https://synojerem.ddns.net/
https://79.88.156.242

enoch85, 2017-06-10:
@ezraholm50 It's in the code already ;) https://github.com/nextcloud/vm/blob/master/lets-encrypt/activate-ssl.sh#L177-L180

ezraholm50, 2017-06-11:
@enoch85 yes when using LE but not when using the snakeoil certs.

newkindofworld, 2017-06-12:
Tried the commands, still not working.
But I guess we're missing the point here, I can't even access nextcloud locally using local Pi address : 192.168.x.x , it gives the same error, shouldn't it be accessible locally ?

enoch85, 2017-06-12:
@newkindofworld Please start over, download the latest image and run it from scratch. Also install SSL with Let's Encrypt when you get promted to do that. Then let us know the outcome.

ezraholm50, 2017-06-12:
You are able to access SSH right? Please post the output of: cat /etc/apache2/sites-available/synojerem.ddns.net.conf

newkindofworld, 2017-06-12:
Started over, everything runs fine now :)

Thanks again !

enoch85, 2017-06-12:
Thanks for confirming.

closed 2017-06-12 by enoch85