techandme/NextBerry #100: Nextberry letsencrypt certificate installation didnt work

FMstar, 2017-06-09:
## Steps to reproduce
1.following the installation script for NEXTBERRY
2.letsencrypt certificate installation fails

## Expected behaviour
certificates should be installed

## Actual behaviour
after reverting setting installation script moves on and installt nextclod without an signed cerificate 

## Server configuration

**Server version:** 
Nextcloud 12.0.0
PHP
Version: 7.0.19
Arbeitspeichergrenzwert: 268435456

Datenbank
Art: mysql
Version: 5.5.54
Größe: 2.1 MB

**Raspberry 2 or 3?**
Raspberry 3

**NextBerry version**
      _  _            _    ___                       __   __ _     ____
       | \| | ___ __ __| |_ | _ ) ___  _ _  _ _  _  _  \ \ / // |   |__ /
       | .` |/ -_)\ \ /|  _|| _ \/ -_)| '_|| '_|| || |  \ V / | | _  |_ \
       |_|\_|\___|/_\_\ \__||___/\___||_|  |_|   \_, |   \_/  |_|(_)|___/
                                                 |__/
 https://www.techandme.se - Nextcloud:v12.0.0 - Uptime: 0 days, 01h13m43s
 ===============================================================================                                           
 RPI: temp=59.1'C - CPU freq: 1400000 - volt=1.3940V - MEM: gpu=32M arm=976M
 ===============================================================================                                           
 Operating system: Raspbian GNU/Linux 8.0 (jessie) (GNU/Linux 4.9.24-v7+ armv7l)   

**Updated from an older installed version or fresh install:**
fresh install

## Network
**Do you use DHCP?**
yes
**Is port 80 and/or 443 open?**
443 open
## Logs / Screenshots

#### Log file (/var/ncdata/nextcloud.log)
```
Insert your nextcloud.log file here
```

#### Installation log (command output)
```
Insert the CLI output here
Do you want to install SSL? ([y]es or [N]o):Y
after checking that my domain is reachable:
Installing letsencrypt...
**/var/scripts/activate-ssl.sh:** **line 151: add-apt-repository: command not found_**
[\] Please wait... 
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
[|] Please wait...  
[|] Please wait... 
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 letsencrypt : Depends: certbot but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages were automatically installed and are no longer required:
  libblas-common libblas3 libgfortran3 liblinear1 libpcap0.8 ndiff python-lxml
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
/etc/apache2/sites-available/myprivatedomain.de.conf was successfully created
Site 000-default already disabled
/var/scripts/activate-ssl.sh: line 232: letsencrypt: command not found
Enabling site 000-default.
To activate the new configuration, you need to run:
  systemctl reload apache2
[0;96mIt seems like no certs were generated, we will do 2 more tries.[0m
[0;32mPress any key to continue...[0m
/var/scripts/activate-ssl.sh: line 244: letsencrypt: command not found
[0;96mIt seems like no certs were generated, we will do 1 more try.[0m
[0;32mPress any key to continue...[0m
/var/scripts/activate-ssl.sh: line 252: letsencrypt: command not found
[0;96mIt seems like no certs were generated, we will do 0 more tries.[0m
[0;32mPress any key to continue...[0m
[0;96mSorry, last try failed as well. :/[0m
3.finally:
[0;32mPress any key to revert settings and exit... [0m

#### Screenshoots
<!--
Please use http://imgur.com/ for screenshots. Thanks!
-->

ezraholm50, 2017-06-10:
Thank you this seems like a scripting error on amd repo vs arm. Fixing this...

ezraholm50, 2017-06-10:
Could you run this: 

sudo apt-get install python-certbot-apache -t jessie-backports

Then: 

sudo bash /var/scripts/
activate-ssl.sh
Please post the output

FMstar, 2017-06-10:
Thank you, but unfortunatelly not sucessfull: 

ncadmin@nextberry:~$ sudo apt-get install python-certbot-apache -t jessie-backports
[sudo] password for ncadmin:
Reading package lists... Done
E: The value 'jessie-backports' is invalid for APT::Default-Release as such a release is not available in the sources

I also ran the activate.ssl script, but as extected with the same errors:
Installing letsencrypt...
/var/scripts/activate-ssl.sh: line 151: add-apt-repository: command not found

ezraholm50, 2017-06-10:
We'll get this working, try this:

sudo apt-get install python-certbot-apache -y

And if.its succesfull then run activate-ssl again

If that doesnt work well git clone the certbot repo

ezraholm50, 2017-06-10:
There is no repo for rasbian, will have to go that way if we want letsencrypt. Thanks @enoch85

enoch85, 2017-06-10:
OK :+1:

ezraholm50, 2017-06-10:
@FMstar please run:

`sudo rm /var/scripts/activate-ssl.sh`
`sudo wget https://github.com/techandme/NextBerry/raw/beta/lets-encrypt/activate-ssl.sh -P /var/scripts/`
`sudo bash /var/scripts/activate-ssl.sh`

FMstar, 2017-06-10:
...that works quite nice.

On first run ive got an error message that cerbot- directory is not empty, but after renaming it the letsencrypt script semms running well.
...untill:
Enabling site 000-default.
To activate the new configuration, you need to run:
  systemctl reload apache2
After that the system stops responding, no ping and so on.
But this might be an problem of some meore testings, Multiple runs of the script or anything else.

Ill try an installation from the scratch this evening...

FMstar, 2017-06-10:
the last process that i could see is the letsencrypt- process with 99,9 percent cpu consumption.

uups..the headspreader is quite hot on my PI3.  
Ill cool him down.

ezraholm50, 2017-06-10:
Afterwards please run: `ls /etc/letsencrypt/live`
And post the output

enoch85, 2017-06-10:
@FMstar **Don't** enable 000-default. It will fail as there already is a redirect in the host created during the LE script.

Eveything is done automatically, so you shouldn't have to fiddle with anything by yourself. ;)

enoch85, 2017-06-10:
and @ezraholm50 

> ...that works quite nice.

Told you so ;)

ezraholm50, 2017-06-10:
Notice it just now. I disable 000-default by default, might get activated again somewhere.. thanks will figure it out.

FMstar, 2017-06-10:
@ezraholm50 
the ls /etc/letsencrypt/live
prints out my domain. e.g. myprivatdomain.de

Installation from the scretch runs without any errors, my nextcloud is now online.
THX for the 1st class support.

This project is absolutely recommendable,...and will be rcomendet.

ezraholm50, 2017-06-11:
Glad everything worked out. Still need to iron out some flaws. Keeping this open untill everything is solved.

enoch85, 2017-06-11:
@ezraholm50 Which are the flaws?

ezraholm50, 2017-06-11:
I think its still not solved @FMstar ?

Also I want to run the script myself and test it before closing this.

ezraholm50, 2017-06-11:
If it doesnt work please post rhe output of: cat /etc/apache2/sites-available/nextcloud_ssl_domain_self_signed.conf

enoch85, 2017-06-11:
@ezraholm50 We should *not* push to use unsafe certificates. As LE works (proven) that's what should be recomended and used. But sure, would be nice if the self-signed worked as well. Though I think that users would be happy with that and not search for other solutions if that worked, which is bad.

ezraholm50, 2017-06-11:
Got the error, its in activate-ssl.sh. @enoch85 You disable it when LE generates the certs and afterwards enable it. Enabled it gives a 500 error. Thats why i've disabled it by default. 

Newest commit should fix it. You'll be able to remove, redownload and rerun activate-ssl.sh @FMstar

enoch85, 2017-06-11:
@ezraholm50 

> Installation from the scretch runs without any errors, my nextcloud is now online.
THX for the 1st class support.

It's already fixed.

ezraholm50, 2017-06-11:
@FMstar Please verify that lets-encrypt certificates are in place and https is working.

FMstar, 2017-06-11:
@ezraholm50 
Installation from the scretch runs without any errors, my nextcloud is now online.
In my opinion the problem with the letsencrypt-script "activate-ssl.sh" is solved.

My WebInterface presents an valid ssl certificat signet from root-ca letsencrypt. Connection ist encrypted with TLS1.2

ezraholm50, 2017-06-11:
@FMstar that's great to hear! Thanks for helping out with the bugs. Hope you enjoy your NextBerry!

enoch85, 2017-06-11:
@ezraholm50 This goes pretty fast: `openssl dhparam -dsaparam -out "$DHPARAMS" 8192`

ezraholm50, 2017-06-11:
Not really, I used the default line in activate-ssl.sh like your line above and it took nearly 30 to 45 mins.

enoch85, 2017-06-11:
Wow!

ezraholm50, 2017-06-11:
Yeah, 4x @1.5 GHZ but only seems to use 1 core for the creation of the DH param.

Now the only issue to solve is https://github.com/techandme/NextBerry/issues/111

I'm off for a week now. Can be online sporadically...

FMstar, 2017-06-13:
...hmm...
as i told installation works well, all certs o.k....but they must be updated cause of their valitity for only 90 days.
Thats not the problem,but the cronjob witch is doing that (especially the script /var/scripts/letsencryptrenew.sh) quits with an error.
I guess thats why the certbot- components are not installed (or can not be found). (Same issue like in this post when initially creating the certificates).
Right, or should i open a new issue?
Slow down, there are still 88 days left...;-)

ezraholm50, 2017-06-13:
Good catch i forgot to change the commands. Ill create a pr and some commands for you to run later this week.

enoch85, 2017-06-13:
Question: is Let's Encrypt installed with PPA or Git?

ezraholm50, 2017-06-13:
Git and the cron command is not adjusted yet. Easy fix..

FMstar, 2017-07-13:
a few days still left. Have you any fix?

enoch85, 2017-07-13:
Cc @ezaholm

FMstar, 2017-07-13:
dont hurry, i just wantet to get an update. 
Its a few days to 10. September 2017..;-) (validy of the Certificate)

ezraholm50, 2017-07-15:
Try this:

`sed -i 's|! certbot|! cd /etc/certbot; ./letsencrypt-auto|g' /var/scripts/letsencryptrenew.sh; sudo bash /var/scripts/letsencryptrenew.sh`

And please post the output.

FMstar, 2017-08-11:
After a RELAXING Holliday i tryed your hints:

root@bennyundmilka:/var/log/letsencrypt# sed -i 's|! certbot|! cd /etc/certbot; ./letsencrypt-auto|g' /var/scripts/letsencryptrenew.sh; sudo bash /var/scripts/letsencryptrenew.sh
root@bennyundmilka:/var/log/letsencrypt# more /var/log/letsencrypt/cronjob.log
Let's Encrypt FAILED!--2017-06-13_12:25
Let's Encrypt FAILED!--2017-06-25_00:00
Let's Encrypt FAILED!--2017-07-02_00:00
Let's Encrypt FAILED!--2017-07-09_00:00
Let's Encrypt FAILED!--2017-07-16_00:00
Let's Encrypt FAILED!--2017-07-23_00:00
Let's Encrypt FAILED!--2017-08-08_09:30
Let's Encrypt FAILED!--2017-08-08_09:41
Let's Encrypt FAILED!--2017-08-08_09:42
Let's Encrypt FAILED!--2017-08-08_09:46
Let's Encrypt FAILED!--2017-08-08_09:52
Let's Encrypt FAILED!--2017-08-11_11:04

I guess that means that the Update of the certificate still fails.

FMstar, 2017-08-11:
I updatet to 
Server version:
Nextcloud 12.0.0

but same result

ezraholm50, 2017-08-11:
We'll figure this out. Please post output of: sudo cat /var/scripts/letsencryptrenew.sh;

enoch85, 2017-08-11:
@ezraholm50 There are some changes in the main scripts, please update the beta branch and test. Everything should work.

https://github.com/nextcloud/vm/blob/master/lets-encrypt/test-new-config.sh#L46-L62

FMstar, 2017-08-21:
sudo cat /var/scripts/letsencryptrenew.sh
#!/bin/sh
service apache2 stop
if ! cd /etc/certbot; ./letsencrypt-auto renew --quiet --no-self-upgrade > /var/log/letsencrypt/renew.log 2>&1 ; then
        echo "Let's Encrypt FAILED!"--$(date +%Y-%m-%d_%H:%M) >> /var/log/letsencrypt/cronjob.log
        service apache2 start
else
        echo "Let's Encrypt SUCCESS!"--$(date +%Y-%m-%d_%H:%M) >> /var/log/letsencrypt/cronjob.log
        service apache2 start
fi

FMstar, 2017-08-21:
/var/log/letsencrypt# more cronjob.log
Let's Encrypt FAILED!--2017-06-13_12:25
Let's Encrypt FAILED!--2017-06-25_00:00
Let's Encrypt FAILED!--2017-07-02_00:00
Let's Encrypt FAILED!--2017-07-09_00:00
Let's Encrypt FAILED!--2017-07-16_00:00
Let's Encrypt FAILED!--2017-07-23_00:00
Let's Encrypt FAILED!--2017-08-08_09:30
Let's Encrypt FAILED!--2017-08-08_09:41
Let's Encrypt FAILED!--2017-08-08_09:42
Let's Encrypt FAILED!--2017-08-08_09:46
Let's Encrypt FAILED!--2017-08-08_09:52
Let's Encrypt FAILED!--2017-08-11_11:04
Let's Encrypt SUCCESS!--2017-08-13_00:00
Let's Encrypt SUCCESS!--2017-08-20_00:00
Let's Encrypt FAILED!--2017-08-21_12:05

Means on 2017-08-13 cronjob runs with success...

lets take a look:
-rw-r--r-- 1 root root   602 Aug 21 12:05 cronjob.log
-rw-r--r-- 1 root root 34845 Aug 21 12:05 letsencrypt.log
-rw-r--r-- 1 root root 19766 Aug 20 00:00 letsencrypt.log.1
-rw-r--r-- 1 root root 19765 Aug 13 00:00 letsencrypt.log.2
-rw-r--r-- 1 root root  6412 Aug 11 11:04 letsencrypt.log.3
-rw-r--r-- 1 root root  6412 Aug  8 09:52 letsencrypt.log.4
-rw-r--r-- 1 root root  6412 Aug  8 09:46 letsencrypt.log.5
-rw-r--r-- 1 root root  6412 Aug  8 09:42 letsencrypt.log.6
-rw-r--r-- 1 root root  6412 Aug  8 09:41 letsencrypt.log.7
-rw-r--r-- 1 root root 46536 Jun 12 22:00 letsencrypt.log.8
-rw-r--r-- 1 root root     0 Jun 12 21:59 letsencrypt.log.9
-rw-r--r-- 1 root root     0 Aug 21 12:04 renew.log

more letsencrypt.log.2
2017-08-12 22:00:05,175:DEBUG:certbot.main:certbot version: 0.15.0
2017-08-12 22:00:05,177:DEBUG:certbot.main:Arguments: ['--quiet', '--no-self-upgrade']
.
.
.
2017-08-12 22:00:21,545:DEBUG:certbot.log:Exiting abnormally:
Traceback (most recent call last):
  File "/root/.local/share/letsencrypt/bin/letsencrypt", line 11, in <module>
    sys.exit(main())
  File "/root/.local/share/letsencrypt/local/lib/python2.7/site-packages/certbot/main.py", line 743, in main
    return config.func(config, plugins)
  File "/root/.local/share/letsencrypt/local/lib/python2.7/site-packages/certbot/main.py", line 693, in renew
    renewal.handle_renewal_request(config)
  File "/root/.local/share/letsencrypt/local/lib/python2.7/site-packages/certbot/renewal.py", line 436, in handle_renewal_request
    len(renew_failures), len(parse_failures)))
Error: 1 renew failure(s), 0 parse failure(s)

but at 2017-08-08:

more letsencrypt.log.4
2017-08-08 07:52:03,101:DEBUG:certbot.main:certbot version: 0.15.0
2017-08-08 07:52:03,103:DEBUG:certbot.main:Arguments: ['--quiet', '--no-self-upgrade']
2017-08-08 07:52:03,103:DEBUG:certbot.main:Discovered plugins: Plu
.
.
.
t 0x75a0c870>, work_dir=<certbot.cli._Default object at 0x75a0c5d0>)
2017-08-08 07:52:03,335:INFO:certbot.renewal:Cert not yet due for renewal

??

FMstar, 2017-08-21:
started script /var/scripts/letsencryptrenew.sh by hand right now, and the certificate is renewed...!

2017-08-21 10:04:54,350:DEBUG:certbot.storage:Writing new private key to /etc/letsencrypt/archive/bennyundmilka.spdns.de/privkey2.pem.
2017-08-21 10:04:54,351:DEBUG:certbot.storage:Writing certificate to /etc/letsencrypt/archive/bennyundmilka.spdns.de/cert2.pem.
2017-08-21 10:04:54,351:DEBUG:certbot.storage:Writing chain to /etc/letsencrypt/archive/bennyundmilka.spdns.de/chain2.pem.
2017-08-21 10:04:54,352:DEBUG:certbot.storage:Writing full chain to /etc/letsencrypt/archive/bennyundmilka.spdns.de/fullchain2.pem.
2017-08-21 10:05:03,917:DEBUG:certbot.storage:Writing new config /etc/letsencrypt/renewal/bennyundmilka.spdns.de.conf.new.
2017-08-21 10:05:03,930:DEBUG:certbot.renewal:no renewal failures

Ill Update the status on next weekend (automatic run of the renew-script..)

FMstar, 2017-08-31:
cronjob reports:
Let's Encrypt FAILED!--2017-08-27_00:00

and in /var/logs/letsencrypt/letsencrypt.log the last two lines are:

2017-08-31 09:20:07,582:INFO:certbot.renewal:Cert not yet due for renewal
2017-08-31 09:20:07,583:DEBUG:certbot.renewal:no renewal failures

Does this mean that the cronjob reports an error even if the certificate is still valid and is not able to be renewed? I guess the earliest time for a renewal is 10 days before the end of validity?

enoch85, 2017-09-01:
Hmm, that's odd. 

The scripts in the VM are updated, so please check that repo and see what's different, and change accordingly and try again.

ezraholm50, 2017-10-07:
Sorry, to busy with work at the time. could you post the content of `sudo crontab -e`

So by hand it works and it may be a false negative in the logs right?