techandme/NextBerry #110: Installing collabora fails on curl

ezraholm50, 2017-06-11:
plex.waaromzomoeilijk.nl

Just ran activate-ssl.sh and rebooted, got a good cert in browser but curl complains, any idea @enoch85 ?

```
curl: (60) SSL certificate problem: self signed certificate
More details here: http://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
```

ezraholm50, 2017-06-11:
```
root@plex:~# curl -k https://plex.waaromzomoeilijk.nl/status.php
<?xml version="1.0" encoding="iso-8859-1"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
         "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
 <head>
  <title>404 - Not Found</title>
 </head>
 <body>
  <h1>404 - Not Found</h1>
 </body>
</html>

```

enoch85, 2017-06-11:
Do you use LE generated by the VM scripts?

ezraholm50, 2017-06-11:
> Just ran activate-ssl.sh and rebooted, got a good cert in browser but curl complains

enoch85, 2017-06-11:
This is what I get:

```
root@XPS-13:~# curl -k https://plex.waaromzomoeilijk.nl/status.php
{"installed":true,"maintenance":false,"needsDbUpgrade":false,"version":"12.0.0.29","versionstring":"12.0.0","edition":"","productname":"Nextcloud"}root@XPS-13:~# 
```

ezraholm50, 2017-06-11:
Hang on, ```
A potential DNS Rebind attack has been detected.
Try to access the router by IP address instead of by hostname.
```

OPNsense... never mind

closed 2017-06-11 by ezraholm50

enoch85, 2017-06-11:
@ezraholm50 Is that automatically, or did you configure it to block DNS rebind?

ezraholm50, 2017-06-11:
Auto, have no clue how this stuff is resolvable. Can't test collabora on the PI untill solved.

enoch85, 2017-06-11:
Aah ok. Was wondering since I use OPNsense as well. Great feature. :)

Hope you can solve it.