techandme/NextBerry #111: Images not displayed over https

ezraholm50, 2017-06-11:
```
core.js:7 JQMIGRATE: Migrate is installed, version 1.4.0
core.js:3 Refused to load the image 'http://localhost/core/img/logo.svg?v=0' because it violates the following Content Security Policy directive: "img-src 'self' data: blob:".

xa @ core.js:3
plex.waaromzomoeilijk.nl/:1 Refused to load the image 'http://localhost/core/img/logo.svg?v=0' because it violates the following Content Security Policy directive: "img-src 'self' data: blob:".
```

ezraholm50, 2017-06-11:
https://help.nextcloud.com/t/logo-in-top-right-is-blank-login-image-is-blank-cant-seem-to-figure-it-out/13451

No working solution yet

ezraholm50, 2017-06-11:
@FMstar Do you have this issue? The Nextcloud logo and background not being displayed on the login page.

FMstar, 2017-06-11:
No, dont have this issue. Everything ist all right.
But keep in Mind, my uptime is only a couple of hours.
In an other forum i read about some issues with PHP7 and NC12.
"meine Cloud hatte im 24h Dauerbetrieb unter PHP7 und NC12 schon Amnesie und vergass im Browser alles richtig dar zu stellen…)"
this affect crome-Browsers? I use Opera, afaik also using Cromium-engine.
Im still testing the system, at the moment no production sys. (Low load)

ezraholm50, 2017-06-11:
Thank you for your input. We will investigate.