techandme/NextBerry #117: Error when setup of collabora

paarsand, 2017-07-14:
Hello!

Got this while setup of collabora on brand new image, in setup.

Everything else is working great, but collabora obviously failed miserably. as it was not either installed or setup with the correct domain. 

Both my domains are active and working, and i did type them in correctly, i've gotten this to work earlier. 

Got some error about APT not existing or something during setup? 
And if i try to setup the collabora app myself with the correct domain in nextcloud i get "internal server error"  in Norwegian ofcourse, maybe not helping so much.

[image]

[image]

paarsand, 2017-07-14:
Also tried to run the collabora.sh afterwords. But this ended in this:

ncadmin@paarsand:~$ sudo bash collabora.sh
[sudo] password for ncadmin:
collabora.sh: line 7: syntax error near unexpected token `newline'
collabora.sh: line 7: `<!DOCTYPE html>'

ezraholm50, 2017-07-15:
I know what the underlying problem is. Rasbian doesnt have letsencrypt in the repo, i've added manual commands, but not in the collabora script yet. Will change it!

But don't expect much of Collabora since its a RAM eater...

ezraholm50, 2017-07-15:
Try collabora.sh in the BETA branch, should work now.