techandme/NextBerry #120: Failed to upgrade from Nextcloud 15.0.4.9 to 16.0.1

tau-ri, 2019-05-17:
Hi,
I'm using the Nextcloud VM from Daniel witch did a great job so far! Thanky to Tech&me!
While trying to upgrade today to the new version 16 using the update script in /var/scripts/update.sh I got a failure.
I think the first try produces a little failure within php requirements and so the update didn't finished well. So after I updated the php Version and rerun the update, I got this behave:

## Steps to reproduce
1. Update with the update script was sucessfully, but I had to update php as well because of 7.0.x is nolonger supportet by Nextcloud 16.x ( showing a text in webfrontend).
2. I updated php: add-apt-repository ppa:ondrej/php & apt-get install -y php7.3
3. a2dismod php7.0 & a2enmod php7.3 + restart apache.
4. Now the nextcloud instance isn't running, so I tried to rerun the update script to finish it's work: sudo /var/scripts/.update.sh

## Expected behaviour
update should run and brin my Nextcloud online again in versin 16.x

## Actual behaviour
the update now fails with many lines of error.
Maybe causing of an php error or a missconfiguration of php7.3 on my VM...I'm not sure...

## Server configuration
Linux [servername] 4.4.0-78-generic #99-Ubuntu SMP Thu Apr 27 15:29:09 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Ubuntu 16.04.6 LTS (GNU/Linux 4.4.0-78-generic x86_64)

php -v
PHP 7.3.5-1+ubuntu16.04.1+deb.sury.org+1 (cli) (built: May  3 2019 10:00:05) ( NTS )
Copyright (c) 1997-2018 The PHP Group
Zend Engine v3.3.5, Copyright (c) 1998-2018 Zend Technologies
    with Zend OPcache v7.3.5-1+ubuntu16.04.1+deb.sury.org+1, Copyright (c) 1999-2018, by Zend Technologies

**Server version:**
Before last update: 15.0.4.9

**Updated from an older installed version or fresh install:**
Updates origin from (14.x :: not sure if ::) to 15.x to now 16.x

## Network
**Do you use DHCP?**
no!

**Is port 80 and/or 443 open?**
yes

## Logs / Screenshots
This is a bit shortet as there are many repeating lines within the same File/Line number...
SHORTED LOG ----- Means, I removes the repeating error blocks with the same linenumber and text...

ncadmin@serveryxz:/var/scripts$ sudo ./update.sh
[sudo] password for ncadmin:
PHP Fatal error:  Uncaught Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occurred in driver: could not find driver in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(448): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(410): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(354): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(710): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(181): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configur in /var/www/nextcloud/lib/private/DB/Connection.php on line 64
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Trying to upgrade the Redis Pecl extenstion...

Warning: Invalid argument supplied for foreach() in Command.php on line 249
PHP Warning:  Invalid argument supplied for foreach() in /usr/share/php/PEAR/Command.php on line 249
SHORTED LOG -----

Notice: Undefined index: honorsbaseinstall in Role.php on line 173
PHP Notice:  Undefined index: honorsbaseinstall in /usr/share/php/PEAR/Installer/Role.php on line 173
SHORTED LOG -----

Notice: Undefined index: installable in Role.php on line 139
PHP Notice:  Undefined index: installable in /usr/share/php/PEAR/Installer/Role.php on line 139
SHORTED LOG -----

Notice: Undefined index: phpfile in Role.php on line 204
PHP Notice:  Undefined index: phpfile in /usr/share/php/PEAR/Installer/Role.php on line 204
SHORTED LOG -----

Notice: Undefined index: config_vars in Role.php on line 46
PHP Notice:  Undefined index: config_vars in /usr/share/php/PEAR/Installer/Role.php on line 46
SHORTED LOG -----

Warning: Invalid argument supplied for foreach() in PEAR/Command.php on line 249
PHP Warning:  Invalid argument supplied for foreach() in /usr/share/php/PEAR/Command.php on line 249
SHORTED LOG -----

XML Extension not found
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Checking latest released version on the Nextcloud download server and if it's possible to download...
[\] Please wait... SUCCESS!
Latest release is: 16.0.1. Current version is: .
New version available! Upgrade continues...
Backing up files and upgrading to Nextcloud 16.0.1 in 10 seconds...
Press CTRL+C to abort.

tau-ri, 2019-05-17:
I think

closed 2019-05-17 by tau-ri

ezraholm50, 2019-05-17:
Hey, this repo is currently not maintained. It should work on a fresh install but dont try to upgrade.

Hit me up in a week or so, ill have some time free then.

Kind regards / Met vriendelijke groet,

Ezra Holm